FRC-6217/Team2020 #28: Climbing Prototype

kristyallen6217, 2020-01-05:


buchheitriley, 2020-01-14:
Attached are files for a bar clamp system. The idea is that the clamps can adapt to the angle of the bar and the robot will be able to go up vertically aligned. 

[image]

kristyallen6217, 2020-02-18:
Quick proof-of-concept to verify we'll get it working
 Finish installation and final testing

closed 2020-03-17 by josephtcoyle

buchheitriley, 2020-03-18:
https://www.youtube.com/watch?v=8xeX62mLcf8&t=1184s

The link is a video from Sunday and an aborted launch by SpaceX. I
encourage you to watch this to show you that not everything in the world is
affected by the coronavirus. If you watch the preflight information they
say that SpaceX is in need of engineers and welders. Yes welders. That only
requires a two year associate's degree. The preflight launch show last
about 10 minutes, skip around the video if needed. Hope you enjoy it!

Riley Buchheit
9th Grade Science/ STEM 7 Teacher
Cannon Falls Area Schools