FRC-6217/Team2020 #34: Game Strategy during autonomous

kristyallen6217, 2020-01-05:


Zoejesh6217, 2020-01-10:
Auto program 1                      ***Programs 1&3 might require driving before shooting***
Robot A configuration
- shoot at angle
    - with optional delay
- drive away from line toward trench or rendezvous point
?pick up balls? 

Auto program 2
Robot B configuration
- shoot center, try for inner port
   - optional delay
- drive away (same locations) 
?pick up balls?

Auto program 3
Robot C configuration
- shoot at angle
   - optional delay
-drive away (same locations)
?pick up balls?

Zoejesh6217, 2020-01-10:
Auto program 4
Robot configuration A or C
- load balls into center robot
- drive off line and potentially pick up balls

closed 2022-02-16 by josephtcoyle